Ekumen-OS/beluga #232: Timeline mixup when running bagfile

glpuga, 2023-06-19:
### Bug description

A few transforms and messages are getting stamped with unmatched timestamps, probably as a result of a mixup when the bagfile was generated.

As a result of this, amcl is publishing the map->odom tf in a totally different timescale than the odom->base and base->laser_front , causing RViz to fail to display the later.

See timestamps on the logs and on the two transforms on the right:

[image]

See timestamps on the /clock and /scan topics:

[image]

On RViz,

[image]

**Platform (please complete the following information):**
 - OS: Humble development container included in Beluga
 - `Beluga` version: ed7dfe38bcab566a900b636378e7dbf05981120a

### How to reproduce
_List steps to reproduce the issue:_
1. On a console: `cd /ws && colcon build && source install/setup.bash && ros2 launch beluga_example perfect_odometry.launch.xml`
2. On another console: `ros2 topic echo /tf`, `ros2 topic echo /clock`, `ros2 topic echo /scan`
3. On RViz, the TF visor will also show a warning for both base and laser_front.

_Code snippets or minimal examples are always helpful, if not necessary._

**Expected behavior**

All message timestamps to be consistent when running out of a demo and test bagfile.

**Actual behavior**

Timestamps are not consistent with a single timeline, and some information fails to be displayed as a result.

hidmic, 2023-06-19:
I think I saw this when working on #223. Messages carry some simulated clock stamp but `rosbag2` stamped them on arrival using the system clock.

glpuga, 2023-06-20:
This is probably related to `beluga_benchmark` being unable to measure NAV2 AMCL. The node crashes, regardless of the configuration.

Bisecting with git, the problem with AMCL seems to have started starting in #211

nahueespinosa, 2023-06-21:
Ah, that was me.

I did two things that might have caused this:

1. I removed the `/clock` topic messages from the ROS bag.
2. I added the `--clock` option to the ROS bag play command.

The motivation was being able to play ROS bags that don't have `/clock` messages, like the ones we recorded in HQ (_do you have any thoughts on how to do this correctly?_).

Seems like it was not the right approach, we can definitely revert the changes to the ROS bag.

stwirth, 2023-07-07:
From my experience, you always must use `--clock` (with `/clock` topic) to be safe. There are many packages that mix the use of ROS time (which is current system time if no `--clock` is used) and message header stamps.

PR #238 (2023-07-11): Fix `perfect_odometry` rosbag
### Proposed changes

Fixes #232.

This patch fixes message timestamps and starting time in the rosbag to match the header stamps recorded with sim time.

* The contents of the messages were not altered.
* <del>The storage backend was changed to mcap.</del>
* The last messages where there was a collision against one of the walls were deleted.

#### Type of change

- [x] 🐛 Bugfix (change which fixes an issue)
- [ ] 🚀 Feature (change which adds functionality)
- [ ] 📚 Documentation (change which fixes or extends documentation)

### Checklist

- [x] Lint and unit test [... cut]
commit 6e9d7e96 (2023-07-15): Fix `perfect_odometry` rosbag (#238)

Fixes #232.

This patch fixes message timestamps and starting time in the rosbag to
match the header stamps recorded with sim time.

* The contents of the messages were not altered.
* The last messages where there was a collision against one of the walls
were deleted.
closed 2023-07-11 by nahueespinosa