Ekumen-OS/beluga #249: The localization solution in the video in the landing page wobbles on rotations

glpuga, 2023-07-22:
### Bug description

The localization solution seems to wobble around when the robot turns on the video in the landing page in a very noticeable way: 

https://github.com/Ekumen-OS/beluga#overview

I know we already spoke about this, but I don't know what the follow up was and whether we are tracking this issue. It's very noticeable in the video so I'd rather have a tracker issue for this and document follow ups.

**Platform (please complete the following information):**
 - OS: ???
 - `Beluga` version: ???

### How to reproduce
 
Watch the video in https://github.com/Ekumen-OS/beluga#overview

**Expected behavior**

The lidar should stay aligned with the map walls at all times.

**Actual behavior**

It doesn't. When the robot rotates, the lidar loses alignment with the walls in a very noticeable way, it appears to oscillate a bit and then returns to alignment.

### Additional context

When we spoke about this we enumerated a number of possible causes:

- Odometry yaw data issue. This was 99% discarded given the calibration tests that had been performed on the Andino robot used for the capture.
- Visualization artifact due to reproduction speed.
- Insufficient motion model noise to represent the quality of the odometry information available.

hidmic, 2023-11-05:
May want to give this another try once #266 lands.

glpuga, 2023-11-05:
:thinking: