Ekumen-OS/beluga #258: Clustering for pose estimation

hidmic, 2023-09-06:
### Feature description

Connected to https://github.com/Ekumen-OS/beluga/discussions/172. Good ol' AMCL computes sampled mean and covariance from particle clusters, whereas Beluga AMCL computes sampled mean and covariance for the entire posterior particle cloud. Regardless of any arguments in favor or against either approach, Beluga AMCL should offer the choice. We need another estimation mixin.

glpuga, 2023-09-07:
Do we know any paper/source for the implementation of this by AMCL?

hidmic, 2023-09-07:
Not that I know of. That said, the idea of clustering has been floating around for a long while e.g. https://arxiv.org/pdf/cs/0204044.pdf.

hidmic, 2024-07-22:
Addressed as of #275.

closed 2024-07-22 by hidmic