Ekumen-OS/beluga #259: Plugin-based extension of `beluga_amcl`

hidmic, 2023-09-06:
### Feature description

Beluga's design allows for compile-time extension via mixins. This is at odds with most extensible ROS packages, which enable runtime extension via (shared library-based) plugins. It would be beneficial for adoption to enable plugin-based extension of `beluga_amcl`.

### Implementation considerations

We can either make entire particle filters pluggable, or particle filter building blocks by adding "trampoline" mixins. In both cases, interfaces need to be defined. We already have done so for the former.

hidmic, 2024-10-24:
Beluga no longer uses mixins and it is unlikely we'll go down the dynamic plugin path considering Beluga's design. Closing.

closed 2024-10-24 by hidmic

glpuga, 2024-10-25:
I did get asked about this by someone, and my answer was about the same lines.

However, providing a demo/example of how to create a node that does pluginize a pipeline stage (e.g. the sensor model) for a limited input type (e.g. 2D scans) might be valuable to show this is not totally off the cards if people are willing to go this path.