Ekumen-OS/beluga #262: Matplotlib is broken in Humble image, probably others

glpuga, 2023-10-19:
### Bug description

When trying ot run the benchmarking scripts to plot data, this error happens:

[image]

This is a reported issue in Matplotlib: https://github.com/matplotlib/matplotlib/issues/26827

It's important to notice that this issue should affect matplotlib 3.8.0. While the official Ubuntu `python3-package` that is the declared dependency of `beluga_benchmark` installs 3.5.1, a second installation of matplotlib is added to the system through pip when installing `evo==1.21.0` in the docker file, which brings `matplotlib==3.8.0`.

As of now, the only workaround this to manually do:
```
sudo pip3 install matplotlib==3.7.3
```

**Platform (please complete the following information):**
 - OS: Development docker for Humble
 - `Beluga` version: 31445270321482c72dc8a7608385a0682880151b

### How to reproduce
_List steps to reproduce the issue:_
1. Start the Humble docker container
2. Start a python3 terminal
3. Type `import matplotlib.pyplot as plt`

**Expected behavior**

No error output.

**Actual behavior**

The error output shown above is printed.

hidmic, 2023-11-05:
>  a second installation of matplotlib is added to the system through pip when installing evo==1.21.0 in the docker file, which brings matplotlib==3.8.0.

For the time being, let's just peg `matplotlib` to `~3.7.0` in that same Dockerfile and move on.

hidmic, 2024-12-06:
Closing for the same reasons #263 was closed.

closed 2024-12-06 by hidmic