Ekumen-OS/beluga #264: ASan error reported in the AmclNode constructor

agalbachicar, 2023-10-27:
### Bug description

I've run ASan reports via [colcon sanitizer reports](https://github.com/colcon/colcon-sanitizer-reports) over beluga and beluga_amcl. You will be happy to hear that beluga is free of detectable address bugs but there was one caught in beluga_amcl. In particular, I think this is not a problem of AmclNode itself, but a problem in rclcpp that I've also seen in other distros using normal Nodes. The problem involves NodeParameters and other classes... It could be well very well a valid issue or a false positive detection. I've spent some time not so long ago debugging a similar problem in rcl and rclcpp with @ivanpauno, who may remember our interaction, which was also detected by ASan tests.

It could be a good thing to atomize the case to a do-nothing node in a simple test that reproduces the error to make sure it is not related to our code (which I believe it is not related).

**Platform (please complete the following information):**
 - OS: docker, jammy, humble
 - `Beluga` version: https://github.com/Ekumen-OS/beluga/commit/31445270321482c72dc8a7608385a0682880151b

### How to reproduce
_List steps to reproduce the issue:_
1. Follow the steps [here](https://github.com/colcon/colcon-sanitizer-reports) to install the mixins.
2. Compile beluga and beluga_amcl with ASan.
```
$ colcon build --build-base=build-asan --install-base=install-asan --cmake-args -DCMAKE_BUILD_TYPE=Debug --mixin asan-gcc --packages-up-to beluga beluga_amcl --symlink-install --event-handlers console_direct+
```
3. Run tests for each package:
```
$ colcon test --build-base=build-asan --install-base=install-asan --event-handlers sanitizer_report+ --packages-up-to beluga --event-handlers console_direct+
```
and
```
$ colcon test --build-base=build-asan --install-base=install-asan --event-handlers sanitizer_report+ --packages-select beluga_amcl --event-handlers console_direct+
```

The last one produces the following relevant output:

```
test 1                                                                                                   
    Start 1: test_amcl_node                                                                              
                                                                                                         
1: Test command: /usr/bin/python3.10 "-u" "/opt/ros/humble/share/ament_cmake_test/cmake/run_test.py" "/ws
/build-asan/beluga_amcl/test_results/beluga_amcl/test_amcl_node.gtest.xml" "--package-name" "beluga_amcl"
 "--output-file" "/ws/build-asan/beluga_amcl/ament_cmake_gmock/test_amcl_node.txt" "--command" "/ws/build
-asan/beluga_amcl/test/test_amcl_node" "--gtest_output=xml:/ws/build-asan/beluga_amcl/test_results/beluga
_amcl/test_amcl_node.gtest.xml"                                                                          
1: Test timeout computed to be: 60                                                                       
1: -- run_test.py: invoking following command in '/ws/build-asan/beluga_amcl/test':                      
1:  - /ws/build-asan/beluga_amcl/test/test_amcl_node --gtest_output=xml:/ws/build-asan/beluga_amcl/test_r
esults/beluga_amcl/test_amcl_node.gtest.xml                                                              
1: Running main() from gmock_main.cc                                                                     
1: [==========] Running 68 tests from 4 test suites.                                                     
1: [----------] Global test environment set-up.                                                          
1: [----------] 7 tests from TestLifecycle                                                               
1: [ RUN      ] TestLifecycle.FullSpin                                                                   
1: =================================================================                                     
1: ==9720==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x607000012bd0 in thread T0:             
1:   object passed to delete has wrong type:                                                             
1:   size of the allocated type:   72 bytes; 
1:   size of the deallocated type: 1 bytes.                                                    [219/1805]
1:     #0 0x7f5e77d9c24f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_
new_delete.cpp:172                                                                                       
1:     #1 0x55b19386995c in __gnu_cxx::new_allocator<char>::deallocate(char*, unsigned long) /usr/include
/c++/11/ext/new_allocator.h:145                                                                          
1:     #2 0x55b1938568d7 in std::allocator_traits<std::allocator<char> >::deallocate(std::allocator<char>
&, char*, unsigned long) /usr/include/c++/11/bits/alloc_traits.h:496                                     
1:     #3 0x55b19390a1c1 in void rclcpp::allocator::retyped_deallocate<char, std::allocator<char> >(void*
, void*) (/ws/build-asan/beluga_amcl/test/test_amcl_node+0x5351c1)                                       
1:     #4 0x7f5e74cc3dae in rcutils_string_map_fini (/opt/ros/humble/lib/librcutils.so+0x9dae)           
1:     #5 0x7f5e74cf4d05  (/opt/ros/humble/lib/librcl.so+0x22d05)                                        
1:     #6 0x7f5e74cf5027 in rcl_node_resolve_name (/opt/ros/humble/lib/librcl.so+0x23027)                
1:     #7 0x7f5e74cf51f8 in rcl_publisher_init (/opt/ros/humble/lib/librcl.so+0x231f8)                   
1:     #8 0x7f5e751b2a6d in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterfa
ce*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_messa
ge_type_support_t const&, rcl_publisher_options_s const&) (/opt/ros/humble/lib/librclcpp.so+0x158a6d)    
1:     #9 0x7f5e751773bb  (/opt/ros/humble/lib/librclcpp.so+0x11d3bb)                                    
1:     #10 0x7f5e75177af0 in std::_Function_handler<std::shared_ptr<rclcpp::PublisherBase> (rclcpp::node_
interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<c
har> > const&, rclcpp::QoS const&), rclcpp::create_publisher_factory<rcl_interfaces::msg::ParameterEvent_
[... 71 lines cut ...]
1: SUMMARY: AddressSanitizer: new-delete-type-mismatch ../../../../src/libsanitizer/asan/asan_new_delete.
cpp:172 in operator delete(void*, unsigned long)                                                         
1: ==9720==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
1: ==9720==ABORTING                                                                                      
1: -- run_test.py: return code 1                                                                         
1: -- run_test.py: generate result file '/ws/build-asan/beluga_amcl/test_results/beluga_amcl/test_amcl_no
de.gtest.xml' with failed test                                                                           
1: -- run_test.py: verify result file '/ws/build-asan/beluga_amcl/test_results/beluga_amcl/test_amcl_node
.gtest.xml'                                                                                              
1/3 Test #1: test_amcl_node .......................***Failed    0.48 sec 
```

Other tests are successful.

**Expected behavior**

Everything should pass.

**Actual behavior**

There is an error.

### Additional context

N/A

hidmic, 2023-11-05:
This is not on us. `rclcpp`'s use of `std::allocator` for C allocations down in `rcl` and below is broken. [`rclcpp::allocator::retyped_allocate`](https://github.com/ros2/rclcpp/blob/76e2b2677b529329db85ed535ddcac2d045473b9/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L33-L41) specifies a size, but [`rclcpp::allocator::retyped_deallocate`](https://github.com/ros2/rclcpp/blob/76e2b2677b529329db85ed535ddcac2d045473b9/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L59-L68) doesn't (because it can't, because a C++ allocator is being used to implement C malloc / free API). That's what ASan is complaining about.

I would just report upstream and [suppress the warning](https://clang.llvm.org/docs/AddressSanitizer.html#suppressing-reports-in-external-libraries) here.

ivanpauno, 2023-11-07:
> This is not on us. rclcpp's use of std::allocator for C allocations down in rcl and below is broken. [rclcpp::allocator::retyped_allocate](https://github.com/ros2/rclcpp/blob/76e2b2677b529329db85ed535ddcac2d045473b9/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L33-L41) specifies a size, but [rclcpp::allocator::retyped_deallocate](https://github.com/ros2/rclcpp/blob/76e2b2677b529329db85ed535ddcac2d045473b9/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L59-L68) doesn't (because it can't, because a C++ allocator is being used to implement a C malloc / free API). That's what ASan is complaining about.

Lol, I forgot that was a thing.
Yes, that's completely broken, it doesn't really break because `std::allocator` ignores the size when you dealloc AFAIK.
But it can break with custom allocators.