Ekumen-OS/beluga #269: Explore robust pose estimation

hidmic, 2023-12-03:
### Feature description

Recent benchmarks have shown that, while Beluga AMCL and Nav2 AMCL are on par on average, Beluga's worst case performance (i.e. $sup \mathrm{APE}$) is subpar compared to good ol' AMCL.

One possible cause for this discrepancy, and the one I entertain the most, are outliers. Outliers would skew Beluga's pose estimation, which relies on sample statistics only, whereas Nav2 AMCL's clustering algorithms may naturally exhibit some resistance to them.

While we could simply work our way forward with #258, I think there is value in exploring methods purposely designed for outlier rejection, and in particular, those from the robust statistics domain.

### Implementation considerations

Beluga's design is advantageous for this: robust pose estimation support amounts to adding a new estimator mixin.

hidmic, 2023-12-03:
For further reference, section 6.8 of [Robust Statistics: Theory and Methods (with R)](https://onlinelibrary.wiley.com/doi/book/10.1002/9781119214656) book, also co-authored by Ricardo Maronna, covers several of competing methods. I only skimmed through it though.

glpuga, 2023-12-04:
This is ok by me, but we should first test the hyphotesis about what the problem is before looking for performant solutions for it. I would definitely not want spend a couple months implementing an unproven way to solve a problem we are not sure we have.

> While we could rely on clustering for this (e.g. kNN in SE(2) like Nav2 AMCL does, [...]

I would still start with this before trying anything else. This is already far better than what we have and it's been extensively tested in the use cases we are mainly interested in.

Afterwards, let's keep in mind that "real-time" is in the eyes of the beholder. It's the users that strike the balance between their application and the resources they throw into it. Most likely a "slow" solution will not be the default for `beluga_amcl`, but can still be a tool in the `beluga` kit. 

>  Ricardo Maronna, its author, is (or was?) a PI at UNLP 😅. Do you know him @glpuga?

I'm afraid not. Looks like he works at Ciencias Exactas.

glpuga, 2025-02-18:
@glpuga