Ekumen-OS/beluga #279: Improve the library user experience

nahueespinosa, 2024-01-02:
### Feature description

The current user experience for the beluga library is not ideal. Mixins are difficult to understand, use and customize. Complex template instantiation and magic functions (`make_mixin`) are needed to make runtime selection possible. I'd argue that we haven't spent time on user-facing documentation since it's really difficult to come up with a simple example.

I strongly believe this library would be much better if our first class citizens were ranges and range algorithms instead of mixins. They can be composed and customized by design, and we wouldn't need to document basic concepts from scratch, since ranges are part of STL since C++20. This approach would allow us to constraint our algorithms better and even optimize them for particular cases.

As a potential library user, this is the code I would want to be shown first:

```cpp
auto particles = beluga::TupleContainer<int, beluga::Weight>{};
auto motion_model = beluga::DifferentialDrive{...};
auto sensor_model = beluga::LikelihoodField{...};

motion_model.update(...);
sensor_model.update(...);

motion_model.propagate(particles);
sensor_model.reweight(particles);

auto temp_particles = particles | 
                      beluga::views::sample |
                      beluga::views::take_exactly(1000) |
                      ranges::to<TupleContainer>;

std::swap(particles, temp_particles);

auto [mean, covariance] = beluga::estimate(particles);
```

Want to insert random particles while sampling?

```cpp
auto particles = beluga::TupleContainer<int, beluga::Weight>{};
auto motion_model = beluga::DifferentialDrive{...};
auto sensor_model = beluga::LikelihoodField{...};

motion_model.update(...);
sensor_model.update(...);

motion_model.propagate(particles);
sensor_model.reweight(particles);

const auto make_random_state = [&]() { return sensor_model.make_random_state(); };
const auto random_state_probability = ...;

auto temp_particles = particles | 
                      beluga::views::sample |
                      beluga::views::random_intersperse(make_random_state, random_state_probability) |
                      beluga::views::take_exactly(1000) |
                      ranges::to<TupleContainer>;

std::swap(particles, temp_particles);

auto [mean, covariance] = beluga::estimate(particles);
```

Showcasing more complex functionality:

```cpp
auto particles = beluga::TupleContainer<int, beluga::Weight>{};

// Type-erased models
beluga::AnyMotionModel motion_model = beluga::DifferentialDrive{...};
beluga::AnySensorModel sensor_model = beluga::LikelihoodField{...};

motion_model.update(...);
sensor_model.update(...);

// Parallel algorithm support
motion_model.propagate(std::execution::par, particles);
sensor_model.reweight(std::execution::par, particles);

const auto make_random_state = [&]() { return sensor_model.make_random_state(); };
const auto slow_alpha = ...;
const auto fast_alpha = ...;

// State tracking decoupled from the algorithm
auto adaptive_state = beluga::AdaptiveSamplerState{};

const auto min_samples = ...;
const auto max_samples = ...;
const auto kld_epsilon = ...;
const auto kld_z = ...;

auto temp_particles = particles | 
                      beluga::views::adaptive_sample(make_random_state, slow_alpha, fast_alpha, &adaptive_state) |
                      beluga::views::take_while_kld(hasher, min_samples, max_samples, kld_epsilon, kld_z) |
                      ranges::to<TupleContainer>;

std::swap(particles, temp_particles);

auto [mean, covariance] = beluga::estimate(particles);
```

nahueespinosa, 2024-01-02:
@glpuga @hidmic

hidmic, 2024-01-02:
I like the simplicity, provided we have a means to deal with constraints / dependencies between particle filter blocks, provide for algorithms that need random access to the full range of particles (e.g. I would like to see motion and sensor models join the pipeline), enable some degree of particle filter structure configuration in runtime. Also, are there any constraints or requirements on range abstractions that may come back to haunt us later on (e.g. must be copyable, must be movable, etc.)?

nahueespinosa, 2024-01-02:
> I would like to see motion and sensor models join the pipeline

We can borrow the `action` (eager range combinators) concept too:

```cpp
auto particles = beluga::TupleContainer<int, beluga::Weight>{};
auto motion_model = beluga::DifferentialDrive{...};
auto sensor_model = beluga::LikelihoodField{...};

motion_model.update(...);
sensor_model.update(...);

auto temp_particles = particles | 
                      beluga::actions::propagate(&motion_model) |
                      beluga::actions::reweight(&sensor_model) |
                      beluga::views::sample |
                      beluga::views::take_exactly(1000) |
                      ranges::to<TupleContainer>;

std::swap(particles, temp_particles);

auto [mean, covariance] = beluga::estimate(particles);
```

> enable some degree of particle filter structure configuration in runtime

This one is tricky. The user can always split the pipeline as needed to handle runtime conditions, but I'm not sure if that's what you have in mind.

> are there any constraints or requirements on range abstractions that may come back to haunt us later on (e.g. must be copyable, must be movable, etc.)?

Semantic requirements for views are listed here: https://en.cppreference.com/w/cpp/ranges/view#Semantic_requirements
- views are O(1) move constructible, and move assignable
- views are either O(1) copy constructible/assignable or not copy constructible/assignable

This is a good paper: https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2021/p2415r2.html

I don't see big problems here, although I reckon we will need to study each case in more detail and disable copy-construction if needed. I must say though, that the range library version that is distributed with Ubuntu Focal is a bit old now, and some concept checks are outdated (i.e. viewable ranges used to have to be default constructible).

nahueespinosa, 2024-01-06:
> I don't see big problems here, although I reckon we will need to study each case in more detail and disable copy-construction if needed.

Well, the problem is that there is no clean way of implementing move-only views in `range-v3` today. But I think we can manage anyway. :sweat_smile:

nahueespinosa, 2024-01-14:
@hidmic @glpuga 
I've been looking at the AMCL code again and their exact algorithm goes like this:

```cpp
// NOTE: reweight does not max-normalize here

particles |= beluga::actions::propagate(...) | beluga::actions::reweight(...);
const auto avg = beluga::average(particles | beluga::views::weights);
adaptive_probability_estimator.update(avg);  // update running averages of likelihood of samples
particles |= beluga::actions::normalize(avg);  // prevent unbounded growth

if (should_resample) {
  const auto random_state_probability = adaptive_probability_estimator();  // compute random state probability
  particles |= beluga::views::sample |
               beluga::views::random_intersperse(..., random_state_probability) |
               beluga::views::take_while_kld(...) |
               beluga::actions::assign;  // all particles have weight = 1.0
}
```

While ours looks like this:
```cpp
// NOTE: reweight does not max-normalize here

particles |= beluga::actions::propagate(...) | beluga::actions::reweight(...);
const auto max = *ranges::max_element(particles | beluga::views::weights);
particles |= beluga::actions::normalize(max);  // prevent unbounded growth

if (should_resample) {
  const auto avg = beluga::average(particles | beluga::views::weights);
  adaptive_probability_estimator.update(avg);  // update running averages of likelihood of samples
  const auto random_state_probability = adaptive_probability_estimator();  // compute random state probability
  particles |= beluga::views::sample |
               beluga::views::random_intersperse(..., random_state_probability) |
               beluga::views::take_while_kld(...) |
               beluga::actions::assign;  // all particles have weight = 1.0
}
```

There is a subtle difference there that I think makes the adaptive sampler not so effective. Also the avg-normalization instead of max-normalization could make a difference when inserting random particles. In our case, new particles have the best estimates (1.0) compared to the rest of the particles as soon as they enter the set.

hidmic, 2024-01-14:
> I just want to point out that all of these things are easy to overlook with mixins, but become pretty obvious with ranges.

Point taken.

---

Great find @nahueespinosa. I'll look at it later tonight. Now I'm curious what the net effect is on performance :thinking:

hidmic, 2024-01-15:
Hmm, we have a few things wrong. We max-normalize instead of normalizing over the total probability mass ([like AMCL does](https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/amcl/src/amcl/pf/pf.c#L290)) and we only update the exponential filters on resample, as you correctly pointed out. Computing weight averages after normalizing them to probabilities actually makes sense to me, but I reckon we are already far enough from theory for that argument to have any weight (sums of likelihood cubes :eyes:).

In any case, making these fixes right now is somewhat painful. I'm eager for those 10 LOC particle filters using the new `rangesv3` based API :smiley:

nahueespinosa, 2024-01-22:
Leaving an update on what is emerging as the new API. This represents a complete AMCL implementation, from initialization to iteration through the data points.

```cpp
auto motion = beluga::DifferentialDriveModel{...};
auto sensor = beluga::LikelihoodFieldModel{...};

using Particle = std::tuple<Sophus::SE2d, beluga::Weight>;

auto particles = beluga::views::sample(initial_distribution) |                  //
                 beluga::views::transform(beluga::convert<Sophus::SE2d>) |      //
                 ranges::views::transform(beluga::make_from_state<Particle>) |  //
                 ranges::views::take_exactly(params.max_particles) |            //
                 ranges::to<beluga::TupleVector>;

auto should_update = beluga::policies::on_motion(params.update_min_d, params.update_min_a);

auto should_resample = beluga::policies::every_n(params.resample_interval_count) &&
                       beluga::policies::on_effective_size_drop;

auto hasher = beluga::spatial_hash<Sophus::SE2d>{0.1, 0.1, 0.1};

for (auto [measurement, odom] : datapoints) {
  if (!should_update(odom)) {
    continue;
  }

  particles |= beluga::actions::propagate(std::execution::par, motion(odom)) |
               beluga::actions::reweight(std::execution::par, sensor(measurement)) |
               beluga::actions::normalize;

  adaptive_probability_estimator.update(particles);

  if (should_resample(particles)) {
    const auto random_state_probability = adaptive_probability_estimator();

    auto make_random_particle = [&sensor] {
      static thread_local auto engine = std::mt19937{std::random_device()()};
      return beluga::make_from_state<Particle>(sensor.make_random_state(engine));
    };

    particles |= beluga::views::sample |
                 beluga::views::random_intersperse(make_random_particle, random_state_probability) |
                 beluga::views::take_while_kld(
                     hasher, params.min_particles, params.max_particles, params.kld_epsilon, params.kld_z) |
                 beluga::actions::assign;
  }

  const auto [mean, covariance] = beluga::estimate(particles);
}
```

nahueespinosa, 2024-01-28:
Everything seems to be shaping up nicely, the last issue I think we should solve before considering the new API complete is how to separate the map from the sensor model. Maybe the sensor model will hold a reference to an existing map.

We need a way of generating random samples from a map too, example usage would be:

```cpp
beluga::views::random_intersperse(/* sample from map */, random_state_probability)
```

But other use cases will come up.

hidmic, 2024-01-28:
> the last issue I think we should solve before considering the new API complete is how to separate the map from the sensor model.

As I see it, a sensor model can be stateless or stateful. If stateful, that state may be coupled to its inputs (like to a map). I think we should allow for both e.g.:

```c++
auto sensor_model = StatelessSensorModel{};
// ...
particles |= beluga::actions::reweight(sensor_model(measurement, map));
// ...
```

and

```c++
auto sensor_model = StatefulSensorModel{map};
// ...
particles |= beluga::actions::reweight(sensor_model(measurement));
// ...
```

I have no strong opinions on how we handle map lifetimes, so long as we are explicit and not overly restrictive.

> If the map (or a wrapper) ends up being a callable that takes a URNG, then it will be straightforward to use for initializing particles

+1000 to the wrapper, to avoid semantic overload (e.g. having a "map" data structure and a "map" distribution). The wrapper could also allow for different distributions over the same map data structure.

commit 687df5e6 (2024-03-11): Deprecate all the (mixin) things (#330)

### Proposed changes

Closes #279.

#### Type of change

- [ ] 🐛 Bugfix (change which fixes an issue)
- [x] 🚀 Feature (change which adds functionality)
- [ ] 📚 Documentation (change which fixes or extends documentation)

💥 **Breaking change!** I tried using the [[deprecated]] attribute, but
these classes were already subtly broken with the functional form of
sensor and motion models. There is no reason to keep them around.

### Checklist

- [x] Lint and unit tests (if any) pass locally with my changes
- [x] I have added tests that prove my fix is effect [... cut]
closed 2024-03-11 by nahueespinosa