Ekumen-OS/beluga #299: Add landmark-based localization examples

hidmic, 2024-01-30:
### Feature description

As of #268, Beluga features a few sensor models that use landmarks for localization. Alas, we have no example nor demo making use of them -- only unit tests. It would be nice to add a one for each (there are two), even if on a much simplified robot model (aka a box with wheels).

### Implementation considerations

This seems like a good opportunity to turn `beluga_example` into `beluga_examples`, though our 2D simulation in `flatland` must give way to a 3D simulation using Gazebo (or Webots or any rigid body simulator supporting camera sensor models).

glpuga, 2024-01-30:
There are demos now in a branch of the kobuki repo. It's only there because I started from the infrastructure for that repo, but it's actually a different robot, different settings and mostly different worlds.

We need to polish them and create a new public repor for them (we need that for RSS anyway).

https://github.com/ekumenlabs/kobuki_ros2_stack/tree/glpuga/beluga_demo

hidmic, 2024-01-30:
FYI @larodriguez22. This is what we were talking about last time. Sorry for the delay. I'll assign you once I'm allowed to.

hidmic, 2024-02-06:
@glpuga @larodriguez22 considering the soon to be public `beluga-demos` repository already includes landmark-based localization examples, I'm inclined to reassign this ticket, so we can call it a day when `beluga-demos` is out. We can open more tickets targetting specific demos (e.g. one with Andino, one that doesn't use ROS, etc.). In fact, perhaps `beluga-demos` is the right place for such tickets.

hidmic, 2024-05-10:
https://github.com/Ekumen-OS/beluga-demos made it out some time ago, so we can call this done.

closed 2024-05-10 by hidmic