Ekumen-OS/beluga #305: Write tutorial on using Beluga

hidmic, 2024-02-09:
### Feature description

Precisely what the title says. While Beluga's API is well documented, the only reference on how to make use of the library is a ROS node and some tests. That works against adoption. We need a golden example that builds up in complexity, showcasing Beluga's features. We need to guide newcomers on reproducing that example, one step at a time.

nahueespinosa, 2024-03-21:
Following up on the conversation on how to implement the golden example, the idea is to implement the one-dimensional hallway example for Monte Carlo Localization presented in Chapter 8.3 in Thrun et. al. using the Beluga core library without ROS dependencies.

[image]

Here is a list of milestones that can be used as reference:

1. Implement the program skeleton, focusing on the creation of a particle set (states and weights) structure and basic visualization capabilities.
    - This can be serializing to a file and then post-processing in Python, or simply writing the output to the terminal in a way that resembles the example in the book.
    - It is key to represent both the position (state) of the particle and the weight (could be a line's height, a circle's diameter, or something else).
2. Develop the initialization function for the particle set to sample uniformly from a specified interval.
    - See [`std::uniform_real_distribution`](https://en.cppreference.com/w/cpp/numeric/random/uniform_real_distribution) and [`beluga::views::sample`](https://github.com/Ekumen-OS/beluga/blob/143e1ae4c062b9a7f8719bccf138e5eff5b7f5f7/beluga/test/beluga/views/test_sample.cpp#L153).
4. Implement the sensor model capable of reweighting particles based on the map landmarks (doors) and the current position.
    - This can use a simplified version of the landmark model proposed in Chapter 6.6.
    - $w = e ^ {{- d^2} / {2 \sigma}}$ where d: _distance to the door_
    - See [`beluga::sensor`](https://github.com/Ekumen-OS/beluga/blob/main/beluga/include/beluga/sensor.hpp#L28) and [`beluga::actions::reweight`](https://github.com/Ekumen-OS/beluga/blob/main/beluga/test/beluga/actions/test_reweight.cpp#L27)
5. Develop the motion model to update the positions of the particles, including a routine for moving particles at a constant rate in a given direction.
    - See [`beluga::motion`](https://github.com/Ekumen-OS/beluga/blob/main/beluga/include/beluga/motion.hpp#L32) and [`beluga::actions::propagate`](https://github.com/Ekumen-OS/beluga/blob/main/beluga/test/beluga/actions/test_propagate.cpp#L28).
6. Implement a basic resampling step that allows for the selection of particles based on their weights (no need to make it adaptive at this stage, so taking samples up to a fixed size is perfectly fine).

FYI @Alondruck @hidmic @gonzodepedro

hidmic, 2024-07-22:
Addressed as of #365 and #387. @Alondruck are the scripts on #387 used for our tutorial?

closed 2024-07-22 by hidmic