Ekumen-OS/beluga #306: Write tutorial on extending Beluga

hidmic, 2024-02-09:
### Feature description

Precisely what the title says. Despite the fact extensibility is one of Beluga's major selling points, we have zero documentation on how one is supposed to do that. Any canonical dynamic system (particle, pendulum) could be used as a basis to demonstrate how algorithms and models are implemented in Beluga.

Make sure to at least cover:

- [ ] A sensor model implementation
- [ ] A motion model implementation
- [ ] A sampling method implementation
- [ ] An estimate method implementation

hidmic, 2024-07-22:
@nahueespinosa is it time to circle back to #358 already?

nahueespinosa, 2024-07-22:
@hidmic I have no bandwidth for this, but yes, I guess so.

PR #450 (2024-10-25): Add user guide on how to extend Beluga
### Proposed changes

Closes #306 . This patch adds a guide on how to extend Beluga. I covered the fraction of the library that is the most stable regarding patterns and architecture and also the most likely one for a user to extend -- new kinds of particles filters or support abstractions I skipped.

#### Type of change

- [ ] 🐛 Bugfix (change which fixes an issue)
- [ ] 🚀 Feature (change which adds functionality)
- [x] 📚 Documentation (change which fixes or extends documentation)

### Checklist

_Put an `x` in the boxes that apply. This is simply a remi [... cut]
commit 08e682fe (2024-10-25): Add user guide on how to extend Beluga (#450)

### Proposed changes

Closes #306 . This patch adds a guide on how to extend Beluga. I covered
the fraction of the library that is the most stable regarding patterns
and architecture and also the most likely one for a user to extend --
new kinds of particles filters or support abstractions I skipped.

#### Type of change

- [ ] 🐛 Bugfix (change which fixes an issue)
- [ ] 🚀 Feature (change which adds functionality)
- [x] 📚 Documentation (change which fixes or extends documentation)

### Checklist

_Put an `x` in the boxes that apply. This is simpl [... cut]
closed 2024-10-25 by hidmic