Ekumen-OS/beluga #354: Cache CI/CD prerequisites

hidmic, 2024-05-09:
### Feature description

Our [CI/CD workflows](https://github.com/Ekumen-OS/beluga/tree/main/.github/workflows) install the same dependencies on every run, which means longer runtimes and heat somewhere around the world. We should leverage regular containers and GHA caching to reuse them.

### Implementation considerations

Containers make sense for environment and tooling dependencies as those almost never change and we can rebuild on demand. For package dependencies that may change on every PR, package manager caching would work best.

hidmic, 2024-05-09:
FYI @nahueespinosa. Also @JADC362. We could use a containerization expert.

nahueespinosa, 2024-05-09:
FWIW, we have a docker build pipeline: https://github.com/Ekumen-OS/beluga/blob/main/.github/workflows/docker_ci_pipeline.yml

It might just be a matter of re-using those images in regular CI.

nahueespinosa, 2024-10-11:
> which means longer runtimes and heat somewhere around the world

For reference, the install step takes 2 to 3 minutes on CI.

It seems like we can [push our images](https://docs.github.com/en/actions/use-cases-and-examples/publishing-packages/publishing-docker-images#publishing-images-to-github-packages) to [GitHub Packages](https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-container-registry) for free. Should we use that or do we have another docker registry to push these things?