Ekumen-OS/beluga #381: Beluga debians fail to build for Noetic on ARM64

hidmic, 2024-05-22:
### Bug description

**Platform (please complete the following information):**
 - OS: Ubuntu Focal
 - `Beluga` version: 2.0.0

### How to reproduce

Unable to reproduce locally on `aarch64` boxes. So far it only happens on the ROS buildfarm/

See https://build.ros.org/job/Nbin_ufv8_uFv8__beluga_amcl__ubuntu_focal_arm64__binary/.

**Expected behavior**

Build completes successfully.

**Actual behavior**

Build fails with tons of compiler errors.

### Additional context

Beluga's development environment != ROS buildfarm release environment. The devil is in the details, or so they say.

nahueespinosa, 2024-05-23:
### How to reproduce

Useful docs: https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/release_jobs.rst#run-the-release-job-locally

Concrete steps:

1. Clone [`ros_buildfarm`](https://github.com/ros-infrastructure/ros_buildfarm).
    ```bash
    git clone git@github.com:ros-infrastructure/ros_buildfarm.git
    cd ros_buildfarm
    ````
1. Setup a virtual environment.
    ```bash
    python3 -m venv venv
    . venv/bin/activate
    ```
1. Install Python dependencies.
    ```bash
    python3 install empy
    ```
1. Add current version of `ros_buildfarm` to the python path.
    ```bash
    export PYTHONPATH=`pwd`:$PYTHONPATH
    ```
1. Generate the release script.
   ```bash
   mkdir beluga-amcl-release
   ros_buildfarm/scripts/release/generate_release_script.py https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml noetic ufv8 beluga_amcl ubuntu focal arm64 > beluga-amcl-release/run.sh
   ```
1. Run the release script.
   ```bash
   sh beluga-amcl-release/run.sh
   ```

nahueespinosa, 2024-05-23:
Using the steps above, I created a beluga Debian package applying patch #384, which I then used to build a new Debian package for beluga_amcl using the buildfarm release scripts.

I noticed that the tests failed to run, which might be an artifact of the hacks I implemented to get this to work. I'm leaving the logs here JIC:

```bash
echo -- Running tests. Even if one of them fails the build is not canceled.                                                                                                                                        
-- Running tests. Even if one of them fails the build is not canceled.                                                                                                                                             
if [ -f "/opt/ros/noetic/setup.sh" ]; then . "/opt/ros/noetic/setup.sh"; fi && \                                                                                                                                   
dh_auto_test || true                                                                                                                                                                                               
        cd .obj-aarch64-linux-gnu && make -j1 test ARGS\+=-j1                                                                                                                                                      
make[2]: Entering directory '/tmp/binarydeb/ros-noetic-beluga-amcl-2.0.0/.obj-aarch64-linux-gnu'                                                                                                                   
Running tests...                                                                                                                                                                                                   
/usr/bin/ctest --force-new-ctest-process -j1                                                                                                                                                                       
Test project /tmp/binarydeb/ros-noetic-beluga-amcl-2.0.0/.obj-aarch64-linux-gnu                                                                                                                                    
    Start 1: _ctest_beluga_amcl_rostest_test_amcl_nodelet.test                                                                                                                                                     
1/1 Test #1: _ctest_beluga_amcl_rostest_test_amcl_nodelet.test ...***Failed    2.98 sec                                                                                                                            
... logging to /home/buildfarm/.ros/log/rostest-galadriel-1097.log                                                                                                                                                 
[ROSUNIT] Outputting test results to /tmp/binarydeb/ros-noetic-beluga-amcl-2.0.0/.obj-aarch64-linux-gnu/test_results/beluga_amcl/rostest-test_amcl_nodelet.xml                                                     
[Testcase: testtest_amcl_nodelet] ... FAILURE!                                                                                                                                                                     
FAILURE: Test node [beluga_amcl/test_amcl_nodelet] does not exist or is not executable                                                                                                                             
  File "/usr/lib/python3.8/unittest/case.py", line 60, in testPartExecutor                                                                                                                                         
    yield                                                                                                                                                                                                          
  File "/usr/lib/python3.8/unittest/case.py", line 676, in run                                                                                                                                                     
    self._callTestMethod(testMethod)                                                                                                                                                                               
  File "/usr/lib/python3.8/unittest/case.py", line 633, in _callTestMethod                                                                                                                                         
    method()                                                                                                                                                                                                       
  File "/opt/ros/noetic/lib/python3/dist-packages/rostest/runner.py", line 93, in fn                                                                                                                               
    self.fail(message)                                                                                                                                                                                             
  File "/usr/lib/python3.8/unittest/case.py", line 753, in fail                                                                                                                                                    
[... cut ...]
 * RESULT: FAIL                                                                                                                                                                                                    
 * TESTS: 0                                                                                                                                                                                                        
 * ERRORS: 0                                                                                                                                                                                                       
 * FAILURES: 1
```

nahueespinosa, 2024-05-23:
@hidmic Should we create a new release?

hidmic, 2024-05-23:
@nahueespinosa at this point, we should. There is nothing else to try locally.

nahueespinosa, 2024-05-28:
Fixed by #384.

See https://build.ros.org/job/Nbin_ufv8_uFv8__beluga_amcl__ubuntu_focal_arm64__binary/buildTimeTrend