Ekumen-OS/beluga #382: Development containers should build in AARCH64

glpuga, 2024-05-22:
### Feature description

In order to be able to build and debug portability issues, it would be helpful if the development container built in ARM AARCH64 hosts (e.g. Raspberry Pi, Graviton instances, etc.).

This should also be supported by CI, to ensure the continuity of this feature.

### Implementation considerations

The current development container is not far from being usable in AARCH64 as it stands today. A number of fixable issues have already been identified:

- `perf` related ubuntu packages  `linux-cloud-tools-generic`, `linux-tools-common` and `linux-tools-generic` fail to be downloaded during image build in ARM.
- The dockerfile is hardcoded to download an x86 build of `fixuid`. It should switch between the amd64 or arm64 builds depending on the machine architecture reported by `uname`. 
- The docker compose configuration mounts the `dvi` device, which may not be available.

hidmic, 2024-06-23:
> This should also be supported by CI, to ensure the continuity of this feature.

There's a catch though. Github provides no cloud hosted, `aarch64` runners, QEMU virtualization is painfully slow, and AWS Graviton instances are expensive. That leaves us to self-hosted runners on SBCs (e.g. RPi4, Jetson's, etc.), which is entirely feasible but some work to setup.

tulku, 2024-06-24:
> > This should also be supported by CI, to ensure the continuity of this feature.
> 
> There's a catch though. Github provides no cloud hosted, `aarch64` runners, QEMU virtualization is painfully slow, and AWS Graviton instances are expensive. That leaves us to self-hosted runners on SBCs (e.g. RPi4, Jetson's, etc.), which are entirely feasible but some work to setup.

There is an Oracle Cloud free tier that includes ARM Ampere A1 CPUs. It might be hard to find free slots (some time ago it was easy - and I have some services running there and works nicely. I tried not so long ago to get another server and there were no available instances), but you might give it shot.

glpuga, 2024-06-24:
> There's a catch though. Github provides no cloud hosted, `aarch64` runners, QEMU virtualization is painfully slow, and AWS Graviton instances are expensive. That leaves us to self-hosted runners on SBCs (e.g. RPi4, Jetson's, etc.), which are entirely feasible but some work to setup.

As a datapoint, the [kobuki repo's only CI](https://github.com/ekumenlabs/kobuki_ros2_stack/actions) builds the docker image (very much like the one in Beluga) in AARCH64 using QEMU/dockerx, and it takes 32 minutes to run. That's not too bad for something that does not need for every MR.

For the record, the fixes addressing this issue are already in glpuga/fixes_arm_docker

PR #411 (2024-07-09): Fixes to get docker to build in arm64
### Proposed changes

Minor fixes to the docker image to fix #382 

#### Type of change

- [x] 🐛 Bugfix (change which fixes an issue)
- [ ] 🚀 Feature (change which adds functionality)
- [ ] 📚 Documentation (change which fixes or extends documentation)

💥 **Breaking change!** _Explain why a non-backwards compatible change is necessary or remove this line entirely if not applicable._

### Checklist

_Put an `x` in the boxes that apply. This is simply a reminder of what we will require before merging your code._

- [x] Lint and unit tests (if any) pass locall [... cut]
commit 0e12b301 (2024-07-09): Fixes to get docker to build in arm64 (#411)

### Proposed changes

Minor fixes to the docker image to fix #382 

Tested building on Raspberry 4B and Raspberry 5.
closed 2024-07-09 by glpuga