Ekumen-OS/beluga #385: Beluga ARM64 ROS 2 debian builds fail sporadically

hidmic, 2024-05-24:
### Bug description

**Platform (please complete the following information):**
 - OS: Ubuntu Jammy
 - `Beluga` version: 2.0.0

### How to reproduce

No way to reproduce. Only observed in the ROS buildfarm, see https://build.ros2.org/job/Hbin_ujv8_uJv8__beluga_amcl__ubuntu_jammy_arm64__binary/1 or https://build.ros2.org/job/Ibin_ujv8_uJv8__beluga_amcl__ubuntu_jammy_arm64__binary/3.

**Expected behavior**
Builds succeed consistently.

**Actual behavior**
Every other builds fails with:

```
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```

### Additional context

This looks like the compiler getting killed by the OS when close enough to exhaust all system memory. I don't know what resources do ROS buildfarm ARM64 instances have or how many release jobs may be running concurrently at any given time, but at the very least we could inspect if any of our templates is particularly hard to expand e.g. using `-ftemplate-depth=n` to search for them by trial and error.

hidmic, 2024-05-30:
It's starting to happen on AMD64 too, see https://build.ros2.org/job/Hbin_uJ64__beluga_amcl__ubuntu_jammy_amd64__binary/3.

hidmic, 2024-05-30:
I did some minimal digging, reducing `-ftemplate-depth` (maximum number of template expansions). IIUC `gcc` defaults it to 900. I can take it down to 100. Compilation breaks at 50, on seemingly innocent code:

<pre>Starting &gt;&gt;&gt; beluga_amcl
--- stderr: beluga_amcl                              
In file included from <b>/usr/include/range/v3/functional/concepts.hpp:19</b>,
                 from <b>/usr/include/range/v3/iterator/concepts.hpp:27</b>,
                 from <b>/usr/include/range/v3/range/access.hpp:34</b>,
                 from <b>/usr/include/range/v3/view/all.hpp:22</b>,
                 from <b>/ws/install/include/beluga/beluga/algorithm/raycasting/bresenham.hpp:21</b>,
                 from <b>/ws/install/include/beluga/beluga/algorithm/raycasting.hpp:22</b>,
                 from <b>/ws/install/include/beluga/beluga/sensor/beam_model.hpp:23</b>,
                 from <b>/ws/src/beluga/beluga_amcl/src/amcl_node.cpp:58</b>:
/usr/include/range/v3/functional/invoke.hpp: In instantiation of ‘<b>constexpr ranges::detail::iterator_associated_types_base_&lt;ranges::iota_view&lt;int, int&gt;::cursor, true&gt;::iterator_associated_types_base_(const ranges::detail::iterator_associated_types_base_&lt;ranges::iota_view&lt;int, int&gt;::cursor, true&gt;&amp;)</b>’:
<b>/usr/include/range/v3/functional/invoke.hpp:138:40:</b>   recursively required from ‘<b>constexpr ranges::basic_iterator&lt;ranges::adaptor_cursor&lt;ranges::basic_iterator&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;::cursor&gt;, ranges::remove_if_view&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;, ranges::logical_negate&lt;beluga::views::detail::take_evenly_fn::operator()&lt;ranges::iota_view&lt;int, int&gt; &gt;(ranges::iota_view&lt;int, int&gt;&amp;&amp;, std::size_t) const::&lt;lambda(const auto:91&amp;)&gt; &gt; &gt;::adaptor&gt; &gt;::basic_iterator(const ranges::basic_iterator&lt;ranges::adaptor_cursor&lt;ranges::basic_iterator&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;::cursor&gt;, ranges::remove_if_view&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;, ranges::logical_negate&lt;beluga::views::detail::take_evenly_fn::operator()&lt;ranges::iota_view&lt;int, int&gt; &gt;(ranges::iota_view&lt;int, int&gt;&amp;&amp;, std::size_t) const::&lt;lambda(const auto:91&amp;)&gt; &gt; &gt;::adaptor&gt; &gt;&amp;)</b>’
<b>/usr/include/range/v3/functional/invoke.hpp:138:40:</b>   required from ‘<b>constexpr decltype ((F&amp;&amp;)(f)((Args&amp;&amp;(ranges::invoke_fn::operator()::args))...)) ranges::invoke_fn::operator()(F&amp;&amp;, Args&amp;&amp; ...) const [with F = ranges::indirected&lt;&lt;lambda(auto:90&amp;&amp;)&gt; &gt;; Args = {ranges::basic_iterator&lt;ranges::adaptor_cursor&lt;ranges::basic_iterator&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;::cursor&gt;, ranges::remove_if_view&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;, ranges::logical_negate&lt;beluga::views::detail::take_evenly_fn::operator()&lt;ranges::iota_view&lt;int, int&gt; &gt;(ranges::iota_view&lt;int, int&gt;&amp;&amp;, std::size_t) const::&lt;lambda(const auto:91&amp;)&gt; &gt; &gt;::adaptor&gt; &gt;&amp;}]</b>’
<b>/usr/include/range/v3/functional/concepts.hpp:36:5:</b>   required by substitution of ‘<b>template&lt;class ... As&gt; char (&amp; ranges::invocable__requires_(concepts::detail::tag&lt;As ...&gt;*, decltype (&amp; invocable__requires_test_&lt;As ...&gt;)))[1] [with As = {ranges::indirected&lt;&lt;lambda(auto:90&amp;&amp;)&gt; &gt;, ranges::basic_iterator&lt;ranges::adaptor_cursor&lt;ranges::basic_iterator&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;::cursor&gt;, ranges::remove_if_view&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;, ranges::logical_negate&lt;beluga::views::detail::take_evenly_fn::operator()&lt;ranges::iota_view&lt;int, int&gt; &gt;(ranges::iota_view&lt;int, int&gt;&amp;&amp;, std::size_t) const::&lt;lambda(const auto:91&amp;)&gt; &gt; &gt;::adaptor&gt; &gt;&amp;}]</b>’
<b>/usr/include/range/v3/functional/concepts.hpp:43:9:</b>   required from ‘<b>constexpr const bool ranges::invocable&lt;ranges::indirected&lt;&lt;lambda(auto:90&amp;&amp;)&gt; &gt;, ranges::basic_iterator&lt;ranges::adaptor_cursor&lt;ranges::basic_iterator&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;::cursor&gt;, ranges::remove_if_view&lt;ranges::cache1_view&lt;ranges::zip_view&lt;ranges::detail::index_view&lt;long unsigned int, long int&gt;, ranges::iota_view&lt;int, int&gt; &gt; &gt;, ranges::logical_negate&lt;beluga::views::detail::take_evenly_fn::operator()&lt;ranges::iota_view&lt;int, int&gt; &gt;(ranges::io [... cut ...]
<b>/usr/include/range/v3/functional/invoke.hpp:138:40:</b> <font color="#CC0000"><b>fatal error: </b></font>template instantiation depth exceeds maximum of 50 (use ‘<b>-ftemplate-depth=</b>’ to increase the maximum)
  138 |             noexcept(noexcept(<font color="#CC0000"><b>((F&amp;&amp;) f)((Args&amp;&amp;) args...)</b></font>))
      |                               <font color="#CC0000"><b>~~~~~~~~~^~~~~~~~~~~~~~~~~~</b></font>
compilation terminated.
</pre>

If buildfarm machines are already loaded, I can see them crashing on such deep template compilation. FYI @nahueespinosa.

hidmic, 2024-06-03:
I've so far failed to reduce template nesting. I think we'll have to go the explicit instantiation route instead, as you suggested @nahueespinosa, and hope we can cut enough bloat that way.

glpuga, 2024-07-12:
Another datapoint here.

The latest versions of Beluga seem ot have a lot of memory, far more than older versions (start of year). Updating the kobuki stack's beluga version to the latest  from a version from last January turned it into unbuildable in a 8GB raspberry 5 unless the sequential colcon executor is used. Peak memory usage happens when bulding `beluga_amcl`, at around 5-6GB.

The same issue happened in a 16GB x64 desktop shortly after. Again, peak memory usage happened on `beluga_amcl`, with a slightly lower peak memory usage than aarch64, but still enough to crash the system when built concurrently with other packages.

hidmic, 2024-07-12:
@glpuga it's the templates. `clang` does a better job than `gcc`, memory consumption-wise. My attempts to optimize it failed. Explicit template instantiation may help.

hidmic, 2024-07-12:
Got another death on the farm https://build.ros2.org/job/Jbin_uN64__beluga_amcl__ubuntu_noble_amd64__binary. We gotta fix this.

glpuga, 2024-07-15:
I don't think this is a situation that deteriorated over time from start of year, really, but now we have two independent targets in beluga_amcl that get built at the same time, doubling the issue.

A stopgap solution might be to temporarily daisy chain those two targets with cmake so they don't get built at the same time, while we look for a better solution.

glpuga, 2024-07-25:
A current thread with a lot of information about this: https://discourse.ros.org/t/openvdb-compile-size/38790/4

> Each build agent is a compute-optimized “xlarge” instance in EC2 spread between c5.xlarge, c6i.xlarge, or m5.xlarge which all have 4vCPUs and 8GiB of memory.
> 
> Each individual package build is expected to use 1/4th of those resources: 1vCPU and 2GiB of RAM. These are soft limits that are not enforced, but since agents run up to four jobs at a time, packages using more than the available memory will be subject to pressure from the OS.

hidmic, 2024-07-25:
I've been following that thread. It basically confirms what we knew: the allocation build jobs get on buildfarm VM is small-ish for today's standards. 

We need the compiler to use less memory.

nahueespinosa, 2024-09-24:
TIL about https://github.com/mikael-s-persson/templight

nahueespinosa, 2024-10-08:
This line has a template depth of 62: https://github.com/Ekumen-OS/beluga/blob/main/beluga_amcl/src/amcl_node.cpp#L194

```
--- stderr: beluga_amcl
In file included from /opt/ros/humble/include/rclcpp/rclcpp/client.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/any_executable.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executor_options.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executor.hpp:37,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:21,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /opt/ros/humble/include/tf2_ros/tf2_ros/buffer_interface.h:48,
                 from /opt/ros/humble/include/tf2_ros/tf2_ros/buffer.h:42,
                 from /ws/src/beluga/beluga_amcl/src/amcl_node.cpp:33:
/usr/include/c++/11/variant: In instantiation of ‘constexpr const size_t std::__detail::__variant::__tuple_count_v<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<> >’:
/usr/include/c++/11/variant:737:2:   required from ‘struct std::__detail::__variant::__tuple_count<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:727:35:   required from ‘constexpr const size_t std::__detail::__variant::__tuple_count_v<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:737:2:   required from ‘struct std::__detail::__variant::__tuple_count<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:727:35:   required from ‘constexpr const size_t std::__detail::__variant::__tuple_count_v<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:737:2:   required from ‘struct std::__detail::__variant::__tuple_count<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:727:35:   required from ‘constexpr const size_t std::__detail::__variant::__tuple_count_v<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:737:2:   required from ‘struct std::__detail::__variant::__tuple_count<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > >)>, std::function<void(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:727:35:   required from ‘constexpr const size_t std::__detail::__variant::__tuple_count_v<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std::function<void(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > >)>, std::function<void(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void(std::shared_ptr<rclcpp::SerializedMessage>, const rclcpp::MessageInfo&)> > >’
/usr/include/c++/11/variant:737:2:   required from ‘struct std::__detail::__variant::__tuple_count<std::function<void(std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::tuple<std: [... cut ...]
/ws/src/beluga/beluga_amcl/src/amcl_node.cpp:194:60:   required from here
/usr/include/c++/11/variant:727:35: fatal error: template instantiation depth exceeds maximum of 62 (use ‘-ftemplate-depth=’ to increase the maximum)
  727 |       __tuple_count<_Tp, _Tuple>::value;
      |                                   ^~~~~
compilation terminated.
```

The ranges library should be like a walk in the park compared to this :sweat_smile:
https://github.com/ros2/rclcpp/blob/f8aea8cc51116ff27fe29fd170d4f23148e9f709/rclcpp/include/rclcpp/any_subscription_callback.hpp#L912

nahueespinosa, 2024-10-10:
Some concrete findings after poking at the codebase for a while:

1. This line requires 164 MB to compile.
   ```cpp
   template class message_filters::Subscriber<sensor_msgs::msg::LaserScan, rclcpp::Node>;
   ```
2. Adding this without using anything explicitly requires 51 MB to compile.
   ```cpp
   #include <tf2_eigen/tf2_eigen.hpp>
   ```

That's not much, but it can add up. Restructuring our libraries differently might help.

nahueespinosa, 2024-10-11:
Leaving these here for future reference.

https://wiki.debian.org/ReduceBuildMemoryOverhead
https://askubuntu.com/a/785815
https://stackoverflow.com/a/26036300

nahueespinosa, 2024-10-13:
Test object analysis:

[image]

hidmic, 2024-10-14:
What's going on with cluster based estimation? :eyes:

nahueespinosa, 2024-10-14:
> What's going on with cluster based estimation?

Many, many layers of range adaptors. I'm currently refactoring the estimation library to address this. A good part of it comes from the test code itself though.