Ekumen-OS/beluga #420: Investigate behavior of the beam sensor model in off-map conditions

glpuga, 2024-08-06:
### Bug description

While performing benchmarking on a number of datasets, it was noticed that the office1-7 bagfile, when replayed through both nav2 and beluga in a "beam sensor" configuration, generates these trajectories:

[image]

where it can be seen that whlie beluga is off when compared to the ground truth, while nav2 is spot on.

Map:

[image]

Some notes, from the conversation about this online:

- All three times this bagfile was benchmarked the same result was obtained.
- The same bagfile, when processed using a likelihood model displays the correct behavior for both beluga and nav2.
- The issue seems to arise because the initial pose is in a region of the map which is unknown in the map. This is so because the map and the ground truth were both determined from the same input bagfile, in which the lidar is front-facing only and the robot never turns to look backwards.
- When initialized in an unknown region, the initial particle cloud collapses because unknown pixels in the map are treated as obstacles by the beam sensor. Both nav2 and beluga apply the same policy in this respect; however, for reason to determine nav2 shows better resiliency and manages to provide correct poses even in this condition.

**Platform (please complete the following information):**
 - OS: [e.g. Ubuntu Focal]
 - `Beluga` version: [e.g. tag, commit sha]

### How to reproduce
_List steps to reproduce the issue:_

 Execute the OpenLoris/Office1-7 benchmark in the [multi-dataset testbench](https://github.com/Ekumen-OS/lambkin/tree/glpuga/dataset_running_toolset) using lambkin.

### Additional context
_Any other information you think could be meaningful to this issue._

glpuga, 2024-08-06:
This is the behavior for both Beluga and Nav2 in the same dataset, but using the Likelihood model:

[likelihood_both.webm](https://github.com/user-attachments/assets/9ced4427-d7f4-40a8-8fe6-a98538754b8c)

glpuga, 2024-08-06:
This is Beluga using the beam model:
[beam_beluga.webm](https://github.com/user-attachments/assets/a4e254e9-3801-4ca0-86b0-cc75f881dc6c)

glpuga, 2024-08-06:
This is Nav2 using the beam model:

[beam_nav2.webm](https://github.com/user-attachments/assets/aa888b39-c9d7-4e9a-8609-6c83e52e8f9f)

hidmic, 2024-08-08:
> When initialized in an unknown region, the initial particle cloud collapses **because unknown pixels in the map are treated as obstacles by the beam sensor**. Both nav2 and beluga apply the same [...]

If I had to guess, that completely arbitrary(*) choice is the problem. Clustered estimates may be helping Nav2 (as the bulk of the probability mass on motion prediction will be off-map). I would change the beam model to look for occupied cells rather than non-free cells.

(*) Not [exactly arbitrary](https://github.com/ros-navigation/navigation2/blob/46c40cb603bb37afcf82cb5f353d51a104bb6284/nav2_amcl/src/map/map_range.c#L34-L36), but that rationale does not apply to Beluga.

> This is Nav2 using the beam model:

Note the particle cloud in that last video is not Nav2's.

glpuga, 2024-08-26:
I did some data gathering using data gathered using the Robomaster sim, in three situations.

Baseline (full map):

[image]

Robot starting in an unknown patch in the map:

[image]

Robot crossing unknown patches along the way:

[image]

I'll upload the full evo results next, but the findings are that:

- as expected, likelihood model is impervious to unknown patches in the map
- Contrary to the observations that triggered this issue, both nav2_amcl and beluga_amcl fail when navigating on unknown patches. This matches the expectations based on code.
- nav2_amcl does show a bit more resiliency when being initialized on top of a small unknown patch, which is similar to the Openloris dataset scenario, but is not impervious to it.

glpuga, 2024-08-26:
Beam model experiments:

[beam_model_results.tar.gz](https://github.com/user-attachments/files/16753369/beam_model_results.tar.gz)

Likelihood model experiments:

[likelihood_model_results.tar.gz](https://github.com/user-attachments/files/16753371/likelihood_model_results.tar.gz)

glpuga, 2024-08-26:
I think we can remove the "bug" label from this, as everything seems to be working as expected in its design.

It does open the question of whether the OpenLoris Office dataset is useful for us, though, but that's for another forum.

hidmic, 2024-08-26:
> I think we can remove the "bug" label from this, as everything seems to be working as expected in its design.

Fair, if we take the beam model design as a given. But IMO that design doesn't make a lot of sense when you think about it. Why assume unknown space is occupied space? Why pay attention to measurements that land in unknown space?