Ekumen-OS/beluga #424: Add option to remove the adaptiveness to the core filter pipeline

serraramiro1, 2024-08-12:
### Feature description

Some map representations don't allow (at least not in a straightforward, practical way) sampling randomly from the map.
This is the case, just to name one, of sparse voxel grids, where there're no notion of free space, only occupied space.

Given this difficulty, the adaptive part of the AMCL pipeline has limited effect, because of not being able to generate random states from the map.

For these cases, an option where adaptiveness is turned off could be of use, and remove the need of forcefully provide a random state generator.
https://github.com/Ekumen-OS/beluga/blob/7a01f39bc78a6124fe0609637b657da0ded9bc6c/beluga/include/beluga/algorithm/amcl_core.hpp#L108

## Definition of done

Add the feature described above, and make sure this is the option we use for all the nodes where randomly sampling from the map doens't make sense.

As of today, `ndt` and `ndt_3d` variants.

hidmic, 2024-09-03:
Coming back to this, is adaptiveness the problem or is it the use of random state samples to fight sample impoverishment? Good ol' AMCL does both, but it need not, does it? @nahueespinosa

nahueespinosa, 2024-10-11:
It does not need to do both. But also, I don't like that we are providing a core amcl class in the main library. I think we should work on a way to easily define it from the pieces so we don't have to make assumptions or maintain multiple variants of similar things.

The particle filter pipeline is a step towards that, but it does not handle state well, that's why we need a class definition that holds all the pieces. There should be a better way of doing this.

hidmic, 2024-10-14:
I think we can discuss if a class is the best incarnation of a complete particle filtering algorithm with state, also if the core library is the best place for complete particle filtering algorithms, but I do think the project should provide them. Not all filtering algorithms are as simple and easy to understand and implement as (bootstrap) AMCL is.