Ekumen-OS/beluga #474: Humble development container fails to build beluga_vdb when following instructions in DEVELOPING.md

glpuga, 2025-02-14:
### Bug description

The repository fails to build `beluga_vdb` when using the ROS Humble image, doing:
```
. /opt/ros/${ROS_DISTRO}/setup.bash
cd /ws
colcon build --symlink-install
```
as per[ DEVELOPING.md](https://github.com/Ekumen-OS/beluga/blob/main/DEVELOPING.md#workflow) instructions.

[image]

**Platform (please complete the following information):**
 - OS: [e.g. Ubuntu Focal]
 - `Beluga` version: [e.g. tag, commit sha]

### How to reproduce

Start docker humble container, and do:
```
. /opt/ros/${ROS_DISTRO}/setup.bash
cd /ws
colcon build --symlink-install
```

**Expected behavior**

All packages build cleanly.

**Actual behavior**

[image]

### Additional context

glpuga, 2025-02-14:
@hidmic should this package bulid in Humble?

hidmic, 2025-02-14:
It would if OpenVDB was properly packaged on Jammy. It's not. Thus why:

https://github.com/Ekumen-OS/beluga/blob/c91454c125277ebd30565d1bb40b4de8ad3a6579/tools/build-and-test.sh#L33-L35

We can do better than this though. We could build and install it from source.

hidmic, 2025-02-14:
Hmm, after some thought, I think the proper solution here (other than branching, which I'd very much try to avoid) is to:

- Install from OpenVDB from sources for the Jammy container.
- Add instructions on how to install OpenVDB from sources on Jammy (on the `beluga_vdb` README)
- Add a note in DEVELOPING about skipping it unless either of the above are satisfied (using COLCON_IGNORE or else)

pvela2017, 2025-02-15:
> Hmm, after some thought, I think the proper solution here (other than branching, which I'd very much try to avoid) is to:
> 
> * Install from OpenVDB from sources within the Jammy container.
> * Add instructions on how to install OpenVDB from sources on Jammy (in the `beluga_vdb` README)
> * Add a note in DEVELOPING.md about skipping `beluga_vdb` builds unless either of the above are satisfied (using COLCON_IGNORE or else)
> 
> FYI [@pvela2017](https://github.com/pvela2017)

Hi!

I can work on this !

pvela2017, 2025-02-25:
@hidmic 
There is a problem with the `range-v3` lib version for Jammy, the version provided is [0.11.0-2](https://launchpad.net/ubuntu/+source/range-v3) for Jammy. And we are using `range/v3/algorithm/fold_left.hpp` [here](https://github.com/Ekumen-OS/beluga/blob/785646357ac35578702985090655ad0907a2f298/beluga_vdb/include/beluga_vdb/sensor/likelihood_field_model3.hpp#L27).

But [fold_left.hpp](https://github.com/ericniebler/range-v3/blob/0.12.0/include/range/v3/algorithm/fold_left.hpp) was introduced on [version 0.12.0](https://github.com/ericniebler/range-v3/releases/tag/0.12.0) - Noble.

> NEW: Fold algorithms from [P2322](http://wg21.link/P2322) (https://github.com/ericniebler/range-v3/pull/1628), (https://github.com/ericniebler/range-v3/pull/1668).

I'm following these [instructions](https://github.com/ericniebler/range-v3?tab=readme-ov-file#building-range-v3---using-vcpkg) to check if it is possible to update the library. Or is there a better way?

pvela2017, 2025-02-26:
> [@pvela2017](https://github.com/pvela2017) perhaps we can switch it to `ranges::accumulate` in `range/v3/numeric/accumulate.hpp`? That one is available in 0.11.0 and it'd achieve the same.

This is seems to fix the problem, thank you!

hidmic, 2025-03-11:
Addressed by #475.

closed 2025-03-11 by hidmic